Roger-random/ESP_8_BIT_composite #21: setRotation support

emilyvelasco, 2021-12-01:
Would it be possible to support the setRotation function present in the Adafruit GFX library in this library? There might be occasions where someone would need to display an image on a composite video monitor rotated to 90, 180 or 270 degrees.

Roger-random, 2021-12-01:
Should be possible. Brief code review didn't find any immediate dealbreakers, I will know more once I dig deeper.

When I built this library I had a big awkward CRT TV that I wouldn't want to turn on its side (or upside-down) and I selfishly neglected to consider someone else may have a smaller CRT practical for rotation. Oops.

emilyvelasco, 2021-12-01:
Not selfish at all! Most people will probably never turn a CRT TV on its side ever in their lives

Roger-random, 2021-12-02:
It turns out to be pretty straightforward. I've just published `setRotation()` support in v1.3.0.
Version release: #22 
GitHub commit: 8f80471815ffbe2d6cf427962bb8f3591a60d95b

closed 2021-12-02 by Roger-random