Roger-random/ESP_8_BIT_composite #23: Compilation Error

jhonrputra, 2021-12-10:
Hi...

I've tried the library and got this error:

SP_8_BIT_composite.cpp:178:33: error: pal_yuyv causes a section type conflict with _sync_type
 const static DRAM_ATTR uint32_t pal_yuyv[] = {

Any advise, please....?

Regards,
Jhon R Putra

Roger-random, 2021-12-10:
This library only defines `pal_yuyv` once in file `ESP_8_BIT_composite.cpp` for use in the same file. There should be no conflict.

Are you trying the library by itself, or is it part of a project that uses other libraries as well? One potential explanation is that another library in your project has also defined a variable of the same name thus causing a conflict.

nilsheuman, 2021-12-26:
Hi, I had the same issue while testing on a esp32, tried removing `DRAM_ATTR` from line ~562 in `ESP_8_BIT_composite.cpp`
```
uint8_t DRAM_ATTR _sync_type[8] = {0,0,0,3,3,2,0,0};
```
And it worked, maybe not the correct solution but it gave an image on the display I have

Roger-random, 2022-01-08:
I'm not familiar enough with this part of the code to know if this is the correct solution. I will update this issue if I learn more.

Roger-random, 2022-06-28:
I have not seen this compilation error in the past six months. Closing as not reproducible.

If somebody experiences this problem, please feel free to reopen this issue. I ask that you make your Arduino project available on a GitHub repository so I could clone the project and see this error for myself. So I could fix it!

closed 2022-06-28 by Roger-random