Roger-random/ESP_8_BIT_composite #24: Compiler warning: format '%s' expects argument of type 'char*', but argument 2 has type 'int' [-Wformat=] [73, 46]

vandenzel, 2022-01-01:
Hi,

I was trying to use this library for testing, and my compiler warns about this:

https://github.com/Roger-random/ESP_8_BIT_composite/blob/001174d6b526de45c6792b26d53042dcc9f6de22/ESP_8_BIT_composite.cpp#L71-L75

The compiler could be right, as far as I know you can't use `%s` on an `int`.

Compiler: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-97-gc752ad5) 5.2.0 (platformio-ide on vscode, reasonably up-to-date)

Regards, Alex.

Roger-random, 2022-01-07:
I've been working in Arduino IDE and it doesn't seem to surface compiler warnings. I'll try to follow your steps to compile under PlatformIO on VSCode, and see if I see it as well.

vandenzel, 2022-01-07:
It's rather obvious when you look at it: `n` is defined as `int`, and within the `printf` you're using `%s` to format it. It will compile, it will even execute, but tries to format a char * starting at address `n`, which is probably not what you want.

Roger-random, 2022-01-08:
I understand the warning and agree with the technical issue. However, I need to see the warning first. Otherwise I wouldn't be able to verify the warning is fixed!

Roger-random, 2022-06-22:
Finally figured it out: apparently format warnings are not raised by default. I had to add the following line to my platform.ini before I saw the error myself.

`build_flags = -Wformat`

commit 31e8ac95 (2022-06-22): DMA chunk is a number for printf formatting. Fixes #24
closed 2022-06-22 by Roger-random