Roger-random/ESP_8_BIT_composite #26: What version of the Arduino ESP32 core to use?

todbot, 2022-02-28:
Hi,
I've tried this on an Espressif ESP32 Dev Module and it seems to hang in `loop()`. 
I've tried versions 2.0.2, 2.0.1, 1.0.6, and 1.0.4 of the ESP32 Arduino core.
All on Arduino IDE 1.8.19. 

Also, what settings for things like CPU frequency, core, flash speed, etc. are you using?

[image]

Roger-random, 2022-02-28:
Are there any error messages visible on Arduino Serial Monitor?

I'm on Arduino 1.8.19 and ESP32 core 2.0.2.

I don't recall changing any device settings from their defaults, but for comparison here's what I have:

[image]

todbot, 2022-03-01:
No errors, but it appears I cannot run at 80MHz on the ESP-WROOM-32 module from 2016. I get backtraces about not being able to set 80MHz. Changing to 40MHz doesn't crash but also I get no composite video.

Which board are you using?

If I modify `GFX_HelloWorld` to add a few printlns(), like:

```c++
void setup() {
  Serial.begin(115200);
  // Initial setup of graphics library
  videoOut.begin();
  Serial.println("after videoOut.begin");
}
void loop() {
  Serial.println("top of loop");
  // Wait for the next frame to minimize chance of visible tearing
  videoOut.waitForFrame();
  Serial.println("after waitForFrame");
```

the resulting console output on reset is:

```
16:21:22.438 -> ets Jun  8 2016 00:22:57
16:21:22.438 -> 
16:21:22.438 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:21:22.475 -> configsip: 0, SPIWP:0xee
16:21:22.475 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:21:22.475 -> mode:DIO, clock div:2
16:21:22.475 -> load:0x3fff0030,len:1324
16:21:22.475 -> ho 0 tail 12 room 4
16:21:22.475 -> load:0x40078000,len:13480
16:21:22.475 -> ho 0 tail 12 room 4
16:21:22.475 -> load:0x40080400,len:3604
16:21:22.475 -> entry 0x400805f0
16:21:22.579 -> after videoOut.begin
16:21:22.579 -> top of loop
```

with nothing more printed.  So it's hanging on `  videoOut.waitForFrame();` ?

Any tips?

Roger-random, 2022-03-01:
My Amazon purchase history says I bought my first ESP32 dev module in 2019, so I am ignorant about modules from 2016.

My most recent purchase was this board: https://amzn.to/35j77z9 (affiliate link)

I don't know if slower flash memory speed would account for this. All the code needs to run from RAM in order to be fast enough to generate the composite video signal.

Do you know if your module can run ESP_8_BIT in SEGA emulation mode? That's where I extracted the video signal generator from. https://github.com/rossumur/esp_8_bit

todbot, 2022-03-01:
Yeah those are the same ones I have, modulo whatever revisions Espressif have done under the shield.

I did spend some time chasing the hang, and it comes down to [this line](https://github.com/Roger-random/ESP_8_BIT_composite/blob/001174d6b526de45c6792b26d53042dcc9f6de22/ESP_8_BIT_composite.cpp#L575) in `video_sync()`:
```c++
  ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
```

and commenting it out does at least cause the code to continue, at the expense of actually doing the right thing. :)   This goes beyond my meager understanding of FreeRTOS so I'm stuck until I learn more.

todbot, 2022-03-01:
Oh and doing the naive thing and just checking out that rossumur/esp_8_bit repo, opening the .ino, and hitting compile, gives me errors about `'rtc_clk_cpu_freq_get' was not declared in this scope`. So I gave up on that.

Roger-random, 2022-03-01:
The other end is [a few lines down](https://github.com/Roger-random/ESP_8_BIT_composite/blob/001174d6b526de45c6792b26d53042dcc9f6de22/ESP_8_BIT_composite.cpp#L638 ). Just below the comment
```
// Signal video_sync() swap has completed
```

This is inside the interrupt service routine `video_isr()` responsible for drawing scanlines. The next step is to determine if the ISR never started, or if it started and couldn't finish a frame for whatever reason. This might be tricky to debug, as `Serial.println()` isn't going to work from inside an ISR.

ESP_8_BIT code puts the ESP32 I2S peripheral to an off-label use, and the audio phase locked loop peripheral in **_very_** off-label use. I would not be surprised if an Espressif errata sometime within the past few years had an impact.

todbot, 2022-03-01:
okay I found another ESP32 board, this one is apparently slightly newer. When doing `esptool chip_id` on it, the differences are:
* not functional:  `"Chip is ESP32-D0WDQ6 (revision 0)"`
* functions well: `"Chip is ESP32-D0WDQ6 (revision 1)"`

Before I switched to the newer chip, it looked like on the older one, the ISR was never getting triggered, almost like interrupts weren't getting enabled for it. Maybe there's some chip errata about that somewhere, but clearly this isn't a problem with your code. And thank you for it!

closed 2022-03-01 by todbot

Roger-random, 2022-03-01:
Glad to hear you've got it working. And we all learned something! I've added these lessons to README for others in the future.