Roger-random/ESP_8_BIT_composite #40: How switch off the display

brennomessana, 2022-11-08:
HI Roger thank you for you great job, I need to switch off the display after some time, there are some function for do that ?

Roger-random, 2022-11-08:
It is not possible to control power on/off/standby via composite video signal.

If I am wrong about that, please point me to an example and I would love to learn something new and maybe incorporate into this library.

To solve the stated problem, I suggest using an additional ESP32 GPIO pin to control display power. Perhaps something like Adafruit's controllable power switch? https://www.adafruit.com/product/2935

brennomessana, 2022-11-09:
Many Thanks Roger I'll look for a alternative have a good job

Roger-random, 2022-11-09:
You're welcome!

closed 2022-11-09 by Roger-random