Roger-random/ESP_8_BIT_composite #41: Support for external DAC

willfurnell, 2023-01-04:
It appears that the newer ESP32 boards (S2/S3 etc.) no longer have an on board DAC :( 
Do you know if it would be possible to support an external DAC instead?
Thanks!

Roger-random, 2023-01-04:
Minor correction: ESP32-S2 has a DAC and should work. However, you are correct that many other newer variations (S3, C2, C3) do not.

My personal opinion is that if someone is going to go through the work of using an external component, they'll be better off using a component designed specifically to generate composite video signal.

But that wouldn't stop those who want to try doing it for the sake of proving it could be done. Given how creative people have been in hacking ESP32, I can confidently say "Yes, it is possible to support an external DAC instead."

However, I can also confidently say I do not know how to do it myself. :)

For today I'll tag this as "Out of scope" (wontfix) on account of (1) if an external component is involved, people more likely to use something else and (2) my own skill level. If either of those two factors should change in the future, I will gladly revisit.

willfurnell, 2023-02-15:
That is fair enough! Thanks :)

closed 2023-02-15 by willfurnell