Roger-random/ESP_8_BIT_composite #43: NTSC mode color not-correct

kurogedelic, 2023-10-17:
I am currently working with the NTSC mode and have come across an issue with the color palette. Despite implementing the NTSC mode, the colors do not appear as expected. I have attached an image that showcases the results of the following code:

```
videoOut.fillRect(0, 0, 35, 180, 0xFFFF);
videoOut.fillRect(35, 0, 35, 180, 0x001F);
videoOut.fillRect(70, 0, 35, 180, 0xF800);
videoOut.fillRect(105, 0, 35, 180, 0x07E0);
videoOut.fillRect(140, 0, 35, 180, 0x07FF);
videoOut.fillRect(175, 0, 35, 180, 0xF81F);
videoOut.fillRect(210, 0, 35, 180, 0xFFE0);
```

The displayed colors do not seem to match the expected output based on the provided color values.

I don't have a PAL-TV, but if I put the code in PAL mode and watch it on an NTSC-TV, it displays the correct colors with some flickering.
So I think there is a problem specific to the NTSC color palette.

Switching between 8bit/16bit color depth has no particular meaning.

I would appreciate any guidance or assistance in resolving this issue.

Thank you.

[image]

closed 2023-10-17 by kurogedelic

kurogedelic, 2023-10-17:
The colors were wrong on my TV, but the colors were correct on the EPSON projector. It might be a "signal" issue.
[image]

Roger-random, 2023-10-17:
NTSC is infamous for inaccurate colors. An old joke is the acronym stands for Never The Same Color. (*) Thank you for posting a picture to illustrate the issue you saw. I agree the colors were wrong, but they are within range of variation I have seen from NTSC.

Obviously, fundamental issues with NTSC color accuracy are not something I could fix in code.

(*) https://en.wikipedia.org/wiki/NTSC#Comparative_quality

kurogedelic, 2023-10-18:
thanks.
I see. I will buy the new PAL CVBS-HDMI converter!