Roger-random/ESP_8_BIT_composite #44: Example does not work

JLBCS, 2023-11-27:
I am the stupid user.
I just discover theis app after failure to migrate TVout from arduino uno to arduino nano ESP32.
So I downloaded and install ESP_8_bit like indicated and had a try with Hello world. 
Beautiful error message ResolveLibrary(driver/dac.h)
I cannot belive it
What is this mess ?

Roger-random, 2023-11-28:
`driver/dac.h` is part of ESP32 Arduino Core. If it cannot be resolved, something is wrong in Arduino configuration.
* Is ESP32 Arduino Core installed correctly? (https://github.com/espressif/arduino-esp32)
* Is ESP32 selected as the target device? If Arduino IDE is targed to, for example, the original Arduino Uno, then `driver/dac.h` would fail to resolve because the ATmega328P chip at the heart of an Arduino Uno does not have a DAC.
* Many other possibilities...

JLBCS, 2023-11-28:
Hi,
I checked ESP32 level. seems to be OK. See attached pictures for the environment and the compile log where I could see the level of ESP32: [ESP32@2.0.0]
[image]
[compile_error.txt](https://github.com/Roger-random/ESP_8_BIT_composite/files/13485283/compile_error.txt)

JLBCS, 2023-11-28:
Hello,
I did another test.
On a PC which has not arduino installed, I started from scratch and installed the IDE, 
Then I installed ESP32 and then ESP_8_BIT_Color composite video library which installed the example GFX_HelloWorld.
Same problem driver/dac.h missing.

What else can I do ???????

Roger-random, 2023-11-28:
I haven't touched this code in over a year. You are using much newer versions of software and libraries compared to my last-known-good versions I listed in README. It's possible something has changed.

According to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/dac.html

> Include `dac.h` to use the legacy driver or `dac_oneshot.h`, `dac_cosine.h`, and `dac_continuous.h` to use the new driver. The legacy driver might be removed in the future.

Perhaps they have removed the legacy driver! I will need to check...

EtienneGomide, 2023-11-28:
Hello,

Sorry to interrupt your conversation, but the way I did may help you.

For me the installation of the *ESP_8_BIT Color Composite*, from
Roger-random, works following this way:

   - Enter in the *Boards Manager *and install the "*ESP32 by Espressif
   Systems*"
   - Enter in the *Manage Libraries* and install the "*Adafruit GFX Library*
   "
   - Download the *ESP_8_BIT_composite Library* from Github Roger-random
   and install it using the Arduino IDE

* Now Restart the Arduino IDE

Enter on *Tools * *->* *Board* and select the ESP32 board of your choice. (here
I always use the "ESP32 Dev Module" for my boards)

Now open and try one of the examples from:
File -> Examples  -> ESP_8_BIT Color Composite Video Library  -> ...

- - - - - -

* I always do it this way and it works every time.
* I always use this model of ESP32 board:
[image: Esp32 1.jpg]

JLBCS, 2023-11-30:
I get confused with all the ESP32 board.
Right now I have the Arduino Nano ESP32. All previous tests have been run on this board.
By the way I don't know how to access GPIO25 on the Nano ESP32 board for video support.
I anticipated the impossibility with the Nano board so I ordered ESP32-S3-DevKitC-1 ESP32-S3.
Is it compatible with ESP_8_bit  because designations in Readme.md by Roger mentions only ESP32-D0WD  ESP32-D0WDQ6  and ESP32-PICO-D4.

EtienneGomide, 2023-11-30:
I have never used this Arduino Nano ESP32. I always use that ESP32 model I
show to you in the picture.
Yes, the GPIO25 is necessary for video output of the ESP_8_BIT_Composite
library.

JLBCS, 2023-11-30:
I cannot see your picture...

EtienneGomide, 2023-11-30:
This model: ESP32 - Rev 1
<https://pt.aliexpress.com/item/32813938236.html?spm=a2g0o.productlist.main.45.359drrqxrrqxxh&algo_pvid=aeebb6ac-91a1-47ff-95d7-fd240d37d763&aem_p4p_detail=202311300703173171831102841320000461051&algo_exp_id=aeebb6ac-91a1-47ff-95d7-fd240d37d763-22&pdp_npi=4%40dis%21BRL%2120.80%2116.22%21%21%214.04%21%21%402101e64117013565972951708e6c83%2112000034169167456%21sea%21BR%21912918073%21&curPageLogUid=VXKXfHlfjwiD&search_p4p_id=202311300703173171831102841320000461051_23>

[image: image.png]

JLBCS, 2023-12-02:
Today this is what I did:
Autres questions
How to add ESP32 to Arduino board manager?
Installing ESP32 in Arduino IDE (Windows, Mac OS X, Linux ...
To install the ESP32 board in your Arduino IDE, follow these next instructions:

    In your Arduino IDE, go to File> Preferences.
    Open the Boards Manager. Go to Tools > Board > Boards Manager…
    Search for ESP32 and press install button for the “ESP32 by Espressif Systems“:
    That's it

I still have the same problem #include "driver/dac.h" 
library not found.
Have a good weekend

EtienneGomide, 2023-12-02:
Are you still using the Arduino Nano Esp32? Or a other board model of esp32?

JLBCS, 2023-12-02:
I just received a board ESP32-S1-WROOm-1
I then in tool >> board manager >> esp32>> I selected  ESP32 WROOM-DA
I try to compile GFX_HelloWorld and I get 3 unused variable and the compile aborted....
I try to compile my program which work on a Uno processor and compile on arduino nano ESP32 and it fails to load on the wroom ...
Tools>> Board info returns unknown with a S/N, PID and VID

My background IBM main frame from 1968 till 2006. A world of difference !

Roger-random, 2023-12-03:
I have not been able to reproduce the compilation error.

Here were the steps for my latest test:

1. Start with a computer with Windows 11 23H2, without Arduino.
2. Go to https://www.arduino.cc/en/software
3. Select option "Download Windows Win 10 and newer, 64 bits"
4. Run the Arduino installer, agree to all prompts to install USB drivers.
5. Start Arduino IDE
6. Select Help/About. Verify it is the current version 2.2.1
7. Arduino IDE has a default sketch (with blank `setup()` and `loop()`)
8. Select Tools/Board/Arduino AVR Boards/Arduino Nano
9. Click "Verify". Successful compilation indicates basic Arduino IDE is installed correctly.
10. Go to https://github.com/espressif/arduino-esp32
11. Click on "Installing" link which leads to https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html
12. Following those instructions, open File/Preferences and add https://espressif.github.io/arduino-esp32/package_esp32_index.json to "Additional boards manager URLs"
13. Close and restart Arduino IDE
14. Select Tools/Board/Board Manager
15. Filter search with "esp32"
16. Install "esp32 by Espressif Systems" (current version is 2.0.14)
17. Once installed, close and restart Arduino IDE
18. Still on the default sketch (with blank `setup()` and `loop()`)
19. Select Tools/Board/esp32/ESP32 Dev Module
20. Click "Verify". Successful compilation indicates nothing is broken.
21. Select Sketch/Include Library/Manage Libraries...
22. Filter search with "composite"
23. Install "ESP_8_BIT Color Composite Video Library by Roger Cheng" (current version is 1.3.2)
24. Agree to also install library dependencies "Adafruit_GFX" and "Adafruit BusIO"
25. Close and restart Arduino IDE again. (Possibly paranoid but I wanted to be sure.)
26. File/Examples/ESP_8_BIT Color Composite Video Library/GFX_HelloWorld
27. Click "Verify". Compilation was successful, no error message about `driver/dac.h`
28. Click "Upload", sketch displays as expected on composite TV.