Roger-random/ESP_8_BIT_composite #45: Strange behaviour

JLBCS, 2023-12-10:
Hello,
I have developed a program which work on an ARDUINO Uno, an Arduino nanoesp32 and an ESP32 WROOM 32 board.
The output is an LCD 20x4 which work correctly.
I need to add a large display on a TV screen, so I sniffed
[image]
[image]
 into ESP_8_Bit;
I have been able to run 'Hello World' on an ESP32 WROOM 32 board. I did not change any connection 
I started copying ESPP_8_Bit statements into my code and this turn to a disaster!
Progressively I removed the ESP_8_Bit specific statements to end up with only the begin() statement.

On the attached picture, you can see the full TV screen and part of the screen with the PC screen and the LCD. You can read all you like!
I also added to Hello world all the include that are in my code. Does not re-create the problem.

Roger-random, 2023-12-10:
This behavior indicates something has prevented important code from running at time-critical intervals. I didn't see a link to your project code, so my uninformed guesses are:

The "something" is likely part of the 20x4 LCD interface library.

The "important code" is likely the composite video signal generation [interrupt service routine.](https://github.com/Roger-random/ESP_8_BIT_composite/blob/0dbb8d1429c8709b3d15e66d6111164098375584/ESP_8_BIT_composite.cpp#L577) This code needs to run once for every composite video horizontal scan line. At 60 frames per second and 312 lines (240 visible signal + sync + blanking interval) that's once every 1/(60*312) = 0.0000534 of a second. Not much room for error and typically the first thing to go wrong.

JLBCS, 2023-12-11:
Thanks for your prompt answer.
I removed all references to LCD and had a test: same problem
I have an ISR. I changed from Pin 2 to Pin 5 as a source of interruption: Same problem
I removed the ISR invocation: same problem
I will send the code (600+ lines) splited in several tabs on my PC. Hope reconciliation is OK
[Meridienne_Viller_ino.txt](https://github.com/Roger-random/ESP_8_BIT_composite/files/13636412/Meridienne_Viller_ino.txt)

Roger-random, 2023-12-13:
I see many other libraries in your project, not just the LCD interface. I'm not familiar enough with those libraries to know which one might cause interference.

Roger-random, 2024-01-25:
Closing due to lack of activity. Feel free to reopen if there is new information.

closed 2024-01-25 by Roger-random