Roger-random/ESP_8_BIT_composite #50: video_isr and rtc_clk_apll_enable broken

joshua137k, 2024-05-30:
C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp: In function 'void i2s_intr_handler_video(void*)':
C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:45:55: error: invalid conversion from 'const volatile void*' to 'volatile void*' [-fpermissive]
   45 |         video_isr(((lldesc_t*)I2S0.out_eof_des_addr)->buf); // get the next line of video
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
      |                                                       |
      |                                                       const volatile void*
C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:39:41: note:   initializing argument 1 of 'void video_isr(volatile void*)'
   39 | void IRAM_ATTR video_isr(volatile void* buf);
      |                          ~~~~~~~~~~~~~~~^~~
C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp: In function 'esp_err_t start_dma(int, int, int)':
C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:102:40: error: too many arguments to function 'void rtc_clk_apll_enable(bool)'
  102 |             case 3: rtc_clk_apll_enable(1,0x46,0x97,0x4,2);   break;    // 10.7386363636 3x NTSC (10.7386398315mhz)
      |                     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.h:57,
                 from C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:22:
C:\Users\Joshu\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~
C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:103:40: error: too many arguments to function 'void rtc_clk_apll_enable(bool)'
  103 |             case 4: rtc_clk_apll_enable(1,0x46,0x97,0x4,1);   break;    // 14.3181818182 4x NTSC (14.3181864421mhz)
      |                     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
C:\Users\Joshu\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~
C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:106:28: error: too many arguments to function 'void rtc_clk_apll_enable(bool)'
  106 |         rtc_clk_apll_enable(1,0x04,0xA4,0x6,1);     // 17.734476mhz ~4x PAL
      |         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
C:\Users\Joshu\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~
C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp: In destructor 'ESP_8_BIT_composite::~ESP_8_BIT_composite()':
C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:686:28: error: too many arguments to function 'void rtc_clk_apll_enable(bool)'
  686 |         rtc_clk_apll_enable(false,0x46,0x97,0x4,1);
      |         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Joshu\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~
C:\Users\Joshu\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:688:28: error: too many arguments to function 'void rtc_clk_apll_enable(bool)'
  688 |         rtc_clk_apll_enable(false,0x04,0xA4,0x6,1);
      |         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Joshu\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

RotaruAndrei23, 2024-05-30:
Same problem. Current version and 1.2.0 as well

Roger-random, 2024-05-30:
Confirm I see the same compilation errors after updating to Espressif ESP32 Arduino core 3.0.0 released three days ago. (https://github.com/espressif/arduino-esp32/releases)

This error is deep within code inherited from ESP_8_BIT and it'll take time before I can make a fix. Anyone familiar with fixing Espressif's v3 breaking changes are welcome to submit a PR.

In the meantime, the workaround is to downgrade to Espressif's ESP32 Arduino core v2. (Last verified to work with 2.0.14 though 2.0.17 was last before v3.0 was released.)

Roger-random, 2024-05-30:
I have updated README to reflect this problem.

Roger-random, 2024-07-05:
Thank you for looking! More eyes are always useful. There are enough people out there hacking on ESP32 that I'm sure some solution will eventually surface, but I haven't figured it out either.

per1234, 2024-08-30:
@joshua137k @Roger-random please reopen this issue. Although the addition of a note to the readme (https://github.com/Roger-random/ESP_8_BIT_composite/commit/610ca84626e772c2d231fcc150e204cf55697d50) is a mitigation, it does not resolve the problem. This issue should be left open until the incompatibility is actually resolved.

JLBCS, 2024-08-30:
@per1234 @joshua137k @tin-can-tomatoes @Roger-random 

Bonjour, le titre de ce post doit être changé pour indiqué ESP_8_BIT Color.... et indiquer de faire marche arrière sur le niveau de gestionnaire de planche.

Now I translate to make you undestand that the original title does not reflect the real problem.
Hi, the title of the post must be changed to reflect the real problem which is known, to be in ESP_8BIT_colr... and also mantion the way to downgrade the board manager to level 2.017

The problem is known since May 30 but the description of the issue is hebrew. Definately, there is a communication problem.

Roger-random, 2024-08-30:
I've created #56 as the master issue to track all reports of failures due to v3.x Arduino Core breaking change.

closed 2024-08-30 by Roger-random

ithinu, 2025-01-10:
> Unfortunately there are still warnings about using the legacy DAC driver which is now deprecated. I looked into using the dac_continuous driver instead but couldn't figure out a way.

These errors come from another library. Only esp32 should be installed for that board.