Roger-random/ESP_8_BIT_composite #53: Shifted display

JLBCS, 2024-07-12:
I developed on an ESP32-Wroom, a clock based on GPS. I need to display the time on two devices, an LCD and a TV screen.
I prepare 4 records  containing my output. When time come I shoot my records on the LCD and on the TV screen.
I use <hd44780.h>   and <hd44780ioClass/hd44780_I2Cexp.h> to manage the LCD and <ESP_8_BIT_GFX.h> to manage the TV part, inspired by hello world example.
Both displays are updated every second with exactly the same data.

Code is about 700 lines and requires a GPS module and both display
Everything look fine except the the TV display is one second after the TFT display.
I swapped the display so TV is first but that did not do any change.
When I compare the time on my PC, my phone, DC77, and LCD, all of them are in line.
Any clue ?
I ran some more tests to clearly identify the problem.
assume my code display the time every second. I observed that the LCD time is one second ahead of the TV time so I distorted the TV time and display it every 10 seconds.
```
    lcd.setCursor(0, 0);
    lcd.print(rec_0);

    lcd.setCursor(0, 1); 
    lcd.print(rec_1);
 
  v_ctr = v_ctr + 1;
  if (v_ctr <10) return;
  v_ctr = 0;
  // TV for the next frame to minimize chance of visible tearing
  videoOut.waitForFrame();
    // Clear screen
  videoOut.fillScreen(0);
  videoOut.setTextColor(color);
  videoOut.setTextSize(2);

  videoOut.setCursor(5, 90);
  videoOut.print(rec_0); 
  videoOut.setCursor(5, 130);
  videoOut.print(rec_1);

```
TV display time is 10 seconds late. When LCD time switch to 42, TV time switches to 32.

I also ran another test. 
I added  a statement videoOut.waitForFrame(); at the end . Both times on LCD and on TV where identical but the image was tearing.
This issue is not visible in the example Hello world as the code is continuously looping.

JLBCS, 2024-07-13:
I complemented the testing. The display on LCD and TV is refreshed every 10 seconds, so I can take a picture and attach it.
[image]
We clearly observe that the TV is 10 seconds late,.
videoOut.print(rec_0); is buffering the data which is sent to the screen by instruction videoOut.waitForFrame();

JLBCS, 2024-07-13:
Temporary fix:
**efficient but not really smart:**
I replicated the write statement !

Roger-random, 2024-07-13:
Back and front buffers are swapped during `videoOut.waitForFrame();` Therefore if it is called ten seconds after data is drawn, it is expected behavior for the onscreen display to be ten seconds late.

In your scenario it sounds like the call to `videoOut.waitForFrame();` should be moved immediately after your calls to `print()` so the back and front buffers are swapped immediately after text data is updated.

JLBCS, 2024-07-13:
As you recommended, I moved the videoOut.waitForFrame(); after the call for print(). Works fine. Thank you.
I suggest that you update the '_Hello world_' example which induced me in error.

Roger-random, 2024-07-13:
I agree examples can be updated.

The examples run in a tight loop as fast as possible, so it didn't matter if `waitForFrame()` was at the end of `loop()` or at the beginning. But it matters for scenarios like yours where updates are infrequent, so better to be consistent in a way that works for all scenarios.

closed 2024-08-08 by JLBCS