Roger-random/ESP_8_BIT_composite #54: Error of conversion

Giak88, 2024-08-04:
\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:45:55: error: invalid conversion from 'const volatile void*' to 'volatile void*' [-fpermissive]
   45 |         video_isr(((lldesc_t*)I2S0.out_eof_des_addr)->buf); // get the next line of video
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
      |                                                       |
      |                                                       const volatile void*

JLBCS, 2024-08-07:
Similar problem after updating ESP_8_BIT. 
The Hello world example returns:
C:\Users\jacques\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~

My code was fine before update, now I get a bunch of error messages.

`WARNING: library Timezone claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp: In function 'void i2s_intr_handler_video(void*)':
c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:45:55: error: invalid conversion from 'const volatile void*' to 'volatile void*' [-fpermissive]
   45 |         video_isr(((lldesc_t*)I2S0.out_eof_des_addr)->buf); // get the next line of video
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
      |                                                       |
      |                                                       const volatile void*
c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:39:41: note:   initializing argument 1 of 'void video_isr(volatile void*)'
   39 | void IRAM_ATTR video_isr(volatile void* buf);
      |                          ~~~~~~~~~~~~~~~^~~
c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp: In function 'esp_err_t start_dma(int, int, int)':
c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:102:40: error: too many arguments to function 'void rtc_clk_apll_enable(bool)'
  102 |             case 3: rtc_clk_apll_enable(1,0x46,0x97,0x4,2);   break;    // 10.7386363636 3x NTSC (10.7386398315mhz)
      |                     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.h:57,
                 from c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:22:
C:\Users\jacques\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~
c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:103:40: error: too many arguments to function 'void rtc_clk_apll_enable(bool)'
  103 |             case 4: rtc_clk_apll_enable(1,0x46,0x97,0x4,1);   break;    // 14.3181818182 4x NTSC (14.3181864421mhz)
      |                     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
C:\Users\jacques\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~
c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:106:28: error: too many arguments to function 'void rtc_clk_apll_enable(bool)'
  106 |         rtc_clk_apll_enable(1,0x04,0xA4,0x6,1);     // 17.734476mhz ~4x PAL
      |         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
C:\Users\jacques\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~
c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp: In destructor 'ESP_8_BIT_composite::~ESP_8_BIT_composite()':
c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:686:28: error: too many arguments to function 'void rtc_clk_apll_enable(bool)'
  686 |         rtc_clk_apll_enable(false,0x46,0x97,0x4,1);
      |         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\jacques\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~
c:\Users\jacques\Documents\Arduino\libraries\ESP_8_BIT_Color_Composite_Video_Library\ESP_8_BIT_composite.cpp:688:28: error: too many arguments to function 'void rtc_clk_apll_enable(bool)'
  688 |         rtc_clk_apll_enable(false,0x04,0xA4,0x6,1);
      |         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\jacques\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/soc/esp32/include/soc/rtc.h:242:6: note: declared here
  242 | void rtc_clk_apll_enable(bool enable);
      |      ^~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1`

How to remove the last update ?

JLBCS, 2024-08-17:
I have isolated the problem and found a bypass.
See _Checking / controling the library_ in the forum.
Circumvention use the board manager to install core and board version 2.017 which will replace version 3.0.3

per1234, 2024-08-30:
> See Checking / controling the library in the forum.

To make it possible for people to actually find the forum topic @JLBCS referenced, I'll add a link to it:

https://forum.arduino.cc/t/checking-controling-the-library/1292142

Roger-random, 2024-08-30:
Correct, this is a symptom of Espressif Arduino Core breaking change from v2.x to v3.x and reverting back to v2.x is the workaround. I've created #56 for tracking purposes.

closed 2024-08-30 by Roger-random