Roger-random/ESP_8_BIT_composite #61: Flickering on PAL mode

Pancra85, 2025-07-26:
When doing
` ESP_8_BIT_GFX videoOut(false /* = PAL */, 8 /* = RGB332 color */);`
and drawing, there's up and down flickering output. Doesn't happens to me on NTSC output.

Any idea why that could happen??

Great work on the library!! :)

Pancra85, 2025-07-26:
There is a black frame on the screen with PAL that would be awesome if it could get ridded of.
No frame on NTSC

Roger-random, 2025-07-28:
Unfortunately I do not have an analog PAL tube TV and must depend on others to diagnose PAL issues. Also be aware I have stopped working with ESP32 and have no experience resolving issues introduced by ESP-IDF >= 3.0.

Roger-random, 2026-10-04:
As explained in the project README: I am unlikely to allocate the time necessary to properly update this project to accommodate the very significant changes Espressif made in ESP32 SDK.

closed 2026-10-04 by Roger-random